rcdrones/UPSPACK_V3 #1: Enter Sleep Mode Before Low Battery [Feature Request]

StaticVOiDance, 2020-12-07:
For my application I want to halt my RPi when main power input is interrupted and restart when power is back. Now I must wait until the battery depleted and if I shut down early the RPi will not be rebooted.

How can I put the UPS to sleep mode without running the battery all the way down? Is it possible to increase the safety shutdown voltage so the RPi only runs for a few minutes before the USP will turn off the output and go to sleep mode?

PanoptiDon, 2020-12-12:
> 
> 
> For my application I want to halt my RPi when main power input is interrupted and restart when power is back. Now I must wait until the battery depleted and if I shut down early the RPi will not be rebooted.

This is what I'm looking for also.  How is your python?

StaticVOiDance, 2020-12-13:
> > For my application I want to halt my RPi when main power input is interrupted and restart when power is back. Now I must wait until the battery depleted and if I shut down early the RPi will not be rebooted.
> 
> This is what I'm looking for also. How is your python?

Fine, but I'm not sure this is possible in Python or any other language without a modification to the UPS Firmware. As far as I can tell the UPS will only enter sleep mode when the voltage is low. There is no documented way to signal the UPS to enter sleep mode or change the safety shutdown voltage threshold. Entering sleep mode is required to wake the Pi back up when power is restored otherwise it will be stuck in a halted state. Additionally, it doesn't look like the FW source for the UPS is available, only the bin.

Did you have something different in mind?

IAmOrion, 2022-04-23:
I realise this is an old thread, but I also need this feature.  When the main power is disconnected, I want the UPS to initiate a safe shut down and then sleep itself.  Then once main power is restored, it should power the Pi again

maxmlr, 2023-05-13:
@rcdrones is there any new on this? Adding a go-to-sleep feature is essential for may use cases. Happy to help if you can point me at the sources for the firmware!

IAmOrion, 2023-05-13:
> @rcdrones is there any new on this? Adding a go-to-sleep feature is essential for may use cases. Happy to help if you can point me at the sources for the firmware!

This is basically abandonware.  They don't care sadly.