rcdrones/UPSPACK_V3 #3: Flashing all time and no 5V out

OPSDZ, 2020-12-17:
After one week to use its now stopped and just found it flashing RPI off , I was removed the Bat and try again the same issues 
trying to update to 3.12 not work .

andyk1, 2021-01-23:
Tried this method over and over step by step and it does not work. After installing, the Rpi 4B is unusable. Completly locked up and have to reflash to a fresh bootable copy without installing the USBPACK_V3 software.

organizedkaos, 2021-01-26:
I was able to flash the v3.2 firmware to my v3 ups pack successfully using my pi4 as the uart communicator.  It took a couple of tries but you have to follow the instructions exactly.

After the flash though the firmware is reporting incorrect battery capacity.  I am hoping this will get fixed.

andyk1, 2021-01-28:
Hello!!! and thanks for the response.

I managed to flash the firmware to 3.2 But not getting the responses in the directions. I followed them to a T but after checking later I saw it did flash. Now I am unable to get the /home/pi/UPSPACK_V3/UPS_GUI.py to work. Get getting a [Errno 13] Permission denied: /Dev/ttyAMA0 even though it's turned on and pointing to the correct serial port. I've tried... reflashed card and started from scratch so many times I've given up till a better way comes around. Glad to hear you got your UPS somewhat running... To be honest,  without the software it is working as it should for me and no corrupted SD cards YET.

Best Regards.

Andy.

P.S. This is the reason I have set up my pi. Many years of learning python,  PHP, and CSS...  https://www.nicomaparkwx.com.

organizedkaos, 2021-01-28:
On my setup I have the ups connected to uart5 (pi4) so I had to change the python script to look at ttyAMA1.  Check if you have another ttyAMA* device.

andyk1, 2021-01-30:
I do, I have AMA0 and AMA1. Point me to the line to change if you remember. I have so many things going on I can't remember when I slept last...:)

organizedkaos, 2021-01-30:
Inside UPS_GUI_demo.py

The line:
test = UPS2("/dev/ttyAMA0")

I changed to:
test = UPS2("/dev/ttyAMA1")