rcdrones/UPSPACK_V3 #6: Seeing garbled text from UART

p-riebs, 2021-01-28:
Two V3 pack are showing the same garbled text when connected to UART.

I've setup the raspberry pi based on the instructions.
```
pi@RPi4:~ $ ls -l /dev/ | grep AMA
lrwxrwxrwx  1 root root           7 Jan 25 22:12 serial0 -> ttyAMA0
```

Output:
```
Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyAMA0, 19:30:42

Press CTRL-A Z for help on special keys

ø¥ø¥øâ¥øá                  øø¥ø¥øââ¥øá         øø¥øbø¥ø¥øâ¥øãøøø¥øøø¥øbøâ     øø¥øâ¥øã¥øâ          øø¥øâ¥øâø¥ø¥øâ¥øá          ø¥øâ¥øââøáá  ø¥ø"¥øâø¥ø¥øââ¥ø   øø¥p
øáøáððÎøøøøøâ¥øâøb�����øâøá                                                                                                                                                                               p
øá¥øâ¥øâøøø¥ø"¥â¥øá                                                                                                                                                                                          p
øáðüøøøøø¥øb¥øâáøøø¥øâ¥øâ                                                                                                                                                                             p
ø¥øâ¥øáøøøb¥øâøâ                                                                                                                                                                                            
```

I tried upgraded the firmware to (ups_v32.bin) on one of them. The update wasn't successful I think because it wasn't able to authenticate the garbled output.

Let me know what I can do to fix this or any information needed to help debug the issue.

p-riebs, 2021-01-28:
The second board appears to be working now after restarting it a few times 
```
$ SmartUPS V3.10,Vin NG,BATCAP 94,Vout 5221 $
$ SmartUPS V3.10,Vin NG,BATCAP 94,Vout 5257 $
$ SmartUPS V3.10,Vin NG,BATCAP 94,Vout 5246 $
$ SmartUPS V3.10,Vin NG,BATCAP 94,Vout 5257 $
$ SmartUPS V3.10,Vin NG,BATCAP 94,Vout 5259 $
$ SmartUPS V3.10,Vin NG,BATCAP 94,Vout 5240 $
$ SmartUPS V3.10,Vin NG,BATCAP 94,Vout 5264 $
```
This is running an old version of the firmware and the newer version appears to be able to fix the issue with what I'm talking about.
```
* When using UART interface for communication, it solves the problem that the serial port will randomly garble. (No need to update V3.2 firmware if the project only uses to STA single-wire communication)  
```
I tried fixing the first board with the same issue using the latest firmware, but I was never able to get it to output readable values before I tried the update. Is it possible that you have to restart the board until you get read-able characters to do a firmware upgrade?

p-riebs, 2021-01-28:
The firmware upgrade process requires shutting down and starting the board up again. There's no guarantee that the board will output a valid checksum after a reboot since the output could be all garbled. I'm stuck.

organizedkaos, 2021-01-28:
I wish I could help.  I really want to use (and like) this v3 board.  I am having other issues.  I had no issues communicating with the v3 ups via uart5 on my pi4.

If you have a pi4 then try another uart.

ptrakk, 2022-04-06:
I'm having the same issue: garbled text.. but only when i send anything through the rx lines. i have to reboot the upspack to get text. if i try to upload firmware the communication fails and then after the serial terminal shows garbled. i hooked up my oscilloscope and it is properly sending data to the board over uart, however the board can't seem to receive it. is there a way to flash the new firmware directly to the chip?

ptrakk, 2022-04-06:
The upspack is also transmitting the data through its RX line. I think i got suckered into buying a crappy product.. the main issue i am having is all the "undervoltage detected" warnings when running on the upspack.

ptrakk, 2022-04-06:
Looking into IN-CIRCUIT-PROGRAMMING (ICP) section to flash it manually on page 256 of the datasheet. the novoton site 404'd the datasheet, but i found a copy here: https://robu.in/wp-content/uploads/2019/04/DS_N76E003_EN_Rev1.06.pdf

ptrakk, 2022-04-06:
I'm tired of looking at it, however i just read :

Users should note that 5V and GND must be used with a silicone dupont cable. If a normal Dupont wire is used, it will cause a larger wire loss when the current is higher, thus the low voltage symbol of lightning is observed on the screen on the Pi4.

i figured this would be circumvented using 4 dupont wires (2+ & 2-) as well as usb. but still getting undervoltage issues. perhaps this might be the case. the undervoltage problem is gone if upspack is removed.

newpond, 2022-04-07:
> I'm tired of looking at it, however i just read :
>
> Users should note that 5V and GND must be used with a silicone dupont cable. If a normal Dupont wire is used, it will cause a larger wire loss when [...]

I used mine with a good quality (short) usb cable to start with and then moved to silicon Dupont cables at a later date. I made them is short as possible and used 20AWG silicon cable (from amazon).
Both these work fine for me although previously I did find some usb cables and some thinner gauge cables did create the problem you are seeing.

What else do you have attached to the Pi? Screen, dongles etc?

newpond, 2022-04-07:
> Two V3 pack are showing the same garbled text when connected to UART.
>
> I've setup the raspberry pi based on the instructions.
>
> ```
> pi@RPi4:~ $ ls -l /dev/ | grep AMA
> lrwxrwxrwx  1 root [...]

hi, did you disable BT and so forth as mentioned in the instructions?

Can you. post your boot/config.txt file

newpond, 2022-04-07:
> hi, did you disable BT and so forth as mentioned in the instructions?
>
> Can you. post your boot/config.txt file

HI, can you also try the following

Can you open terminal and try the following

sudo raspi-config

this will open a configuration window

select
interface options
then
serial port

select "no" for the first option (to do with login shell)
and then "yes for the second option (to enable the serial hardware itself)

then reboot if it asks you to and try again connecting with minicom

newpond, 2022-04-07:
> I'm having the same issue: garbled text.. but only when i send anything through the rx lines. i have to reboot the upspack to get text. if i try to upload firmware the communication fails and then [...]

hi, can you try the following
HI, can you also try the following

Can you open terminal and try the following

sudo raspi-config

this will open a configuration window

select
interface options
then
serial port

select "no" for the first option (to do with login shell)
and then "yes for the second option (to enable the serial hardware itself)

then reboot if it asks you to and try again connecting with minicom.

The reason I suggest this is that technically the installation guide in the UPS documentation is wrong. (if you followed that)
This is because when you add the lines
enable_uart=1
dtoverlay=disable-bt
to the config.txt file as stated in point 3 of the installation guide you "automatically" enabled the serial logging to shell function, which may cause this problem. (known issue). The data you are seeing may not be coming from the UPS pack but actually from the internal serial login shell connection.
The documentation should really state only to add the disable-bt line and then to go into Raspi-config and made the changes I mentioned.

newpond, 2022-04-07:
> I used mine with a good quality (short) usb cable to start with and then moved to silicon Dupont cables at a later date. I made them is short as possible and used 20AWG silicon cable (from amazon). [...]

can you also confirm what make PSU you are using (and the voltage and current rating)

ptrakk, 2022-04-07:
> What else do you have attached to the Pi? Screen, dongles etc.

I have a powered usb hub running a creality 3d printer and a campark webcam. The powersupply to the pi itself is a "CanaKit 3.5A USB-C Power Supply with Noise Filter (UL Listed) specially designed for the Raspberry Pi 4 (5-foot cable)". I get the low voltage while running the upspack through it, however it is intermittent, but happens frequently under processing load ( i think is the trigger/not confirmed).. It may be my cheap dupont connectors, i've noticed they are an alloy with copper, but it does it with, not only just the usb port too, but also combined with the header power. 

the powered usb hub has a 12V 3A charger, which i suppose has regulation inside the hub. brand is aukey. 

i've tested 3 usb cables with the same result.

I've found that i can run the pi and charger with separate supplies, then connecting just the upspack to the pi headers, which is working well, however it will probably throttle bad if there is a power loss. This is actually fine for my use case.