rcdrones/UPSPACK_V3 #7: Request: Battery voltage

organizedkaos, 2021-01-31:
The current 3.2 firmware only reports battery voltage as GOOD or NG.  

Would be nice to get actual battery voltage (range 4.2 [full] and below).

organizedkaos, 2021-09-01:
This project is dead.  I recommend people find other hardware items that have actual support.

hoffmakl1961, 2021-12-12:
@organizedkaos
a dynamic shutdown?
What is that necessary for? 

The threshold value for the STA signal must always be so large that a safe shutdown is still possible. 
If the value would be set too small, a safe function is no longer possible.
Conclusion: The default value is well chosen and should not be configurable.  

For the Raspberry a good solution.

newpond, 2021-12-20:
I have modified the cmd line script from UPSPACK_V2 to work with V3 and give a terminal output.

Create a file called something like up_cmdline.py and add the text below

then
run chmod +x on that file to make it executable

hope this is useful to someone.

#!/usr/bin/python3

from upspackv2 import *

test = UPS2("/dev/ttyAMA0")
version,vin,batcap,vout = test.decode_uart()
print("--------------------------------")
print(" UPS Version:"+version)
print("--------------------------------")

i = 1

while True:
version,vin,batcap,vout = test.decode_uart()

print("-%s-" %i)

if vin == "NG":
    print("USB input adapter : NOT connected!")
else:
    print("USB input adapter : connected!")
print("Battery Capacity: "+batcap+"%")
print("UPS Output Voltage: "+vout+" mV")
print("\n")

i = i+1

if i == 10000:
    i = 1

closed 2023-01-31 by organizedkaos