rcdrones/UPSPACK_V3 #8: Request:  Configurable Shutdown Threshold for STA line

organizedkaos, 2021-02-06:
So far my tests showed the STA line being triggered when the battery capacity is reported at 5% or less.

Would be great if we could set this value somewhere.

tomsimmons78, 2021-02-23:
I'd like to add my vote for this feature.

Being able to to a run time would be useful too, so you could say 1 hour or 5% which ever comes first

organizedkaos, 2021-09-01:
This project is dead.  I recommend people find other hardware that has actual support.

ph818, 2021-10-04:
Hi, I am looking for more information about the STA line, and STA single bus protocol in general to understand it better. How does it transmit? Is it over USB, or over one of the header pins you solder on?

hoffmakl1961, 2021-12-12:
@ph818
STA provides the signal for the python program shutdown_check.py.   "GPIO.wait_for_edge(shutdown_pin, GPIO.RISING)"
For a functional test, at least STA and GND must be connected to the Raspberry.

@ organizedkaos
The threshold value for STA must always be large enough that a safe shutdown is still possible. 
If the value would be set too small, a safe function is no longer possible.
Conclusion: The default value is well chosen and should not be configurable.

closed 2023-01-31 by organizedkaos