rcdrones/UPSPACK_V3 #10: communication issues.

judgement535, 2021-02-23:
I am trying to set up the UART software. When I view my serial ports it shows this. 
serial0 ->ttyS0 
serial1  -> ttyAMA0

When I run **sudo minicom -D /dev/ttyAMA0 -b 9600** I see the following image. Am I having issues setting this up because it is connecting to the wrong serial? The walkthrough shows it should be serial0. Is there something I need to do to change this. 
[image]

tomsimmons78, 2021-02-24:
@judgement535 

Did you make the changes in boot config etc mentioned in the readme?  Mine was the other way until I made that change.

Tom

judgement535, 2021-02-24:
So I decided to try the other version of the command line and it seems the os is an older one and that fixed it. I get a readout now.  I seem to be having issue with the demo now through I am running it on retropie and using terminal.

spiderdab, 2021-06-10:
Hi, I have the V3 board, setup all the cables on a raspberry pi 4b with a debian 64bit os with a xenomai kernel.
I've the ttyAMA0 linked to /dev/serial1 instead of serial0.
I've tried:
dtoverlay=disable-bt
enable_uart=1
but no luck.
I also tried to totally disable bt with:
sudo systemctl disable hciuart
...no luck, but bt is not enabled..

tried also with:
dtoverlay=pi3-miniuart-bt
no luck again..

any help?

Finside, 2021-06-30:
Same here:
------------------
I've tried:

```
dtoverlay=disable-bt
enable_uart=1

```
but no luck.
I also tried to totally disable bt with:

`sudo systemctl disable hciuart
`
...no luck, but bt is not enabled..

```
tried also with:
dtoverlay=pi3-miniuart-bt

```
no luck again..

Finside, 2021-07-01:
I found USB -> CH340 t->RS232 COM-port (DB9) converter
and connected to TX, RX, GND of UPSPACK_V3.

Everything is fine - I see some data received from UPSPACK_V3
in a simple soft Terminal Ver. 1.9b.
Cool!
Now need to force Raspberry Pi 4 read this data.

newpond, 2021-12-20:
the issue is probably related to the serial port config.
When you add the line enable uart=1 it automatically turns on serial logging to shell which stops the UPS demo program from working.

open terminal and try the following

sudo raspi-config

this will open a configuration window

select
interface options
then
serial port

select "no" for the first option (to do with login shell)
and then "yes for the second option (to enable the serial hardware itself)

then reboot if it asks you to

and then try running the UPS demo program again.

rgds
Jon