rcdrones/UPSPACK_V3 #9: Pi didn't restart

tomsimmons78, 2021-02-23:
I tested the V3 board last night, after 12 hours the STA signal caused the Pi to shutdown correctly.

At this point the power was about 5% I think, the last light was flashing, the power light was still on.

After the shutdown I plugged the mains supply to the UPS back in, the red light started flashing, power light was still on and over time time the battery level lights starts to climb.

It was my understanding that once there was a certainly amount of charge in the battery, Pi would be restarted, however this didn't happen.  However, the more I think about it, I believe the Pi power light was still lit.  If restarting the Pi is achieved simply by power being re-applied, does that mean that I would need to wait longer, until the UPS actually shuts down from lack of power?

If so, then that's a bit annoying, with nothing drawing power apart from the UPS, it's going to take a long time for that last 5% to be consumed, meaning that the only way to resume on power resumption would be via the power switch.

Tom

tomsimmons78, 2021-02-24:
So as I expected, after allowing the UPS to shutdown the Pi (as it had before), and leaving it until the UPS lights were all off, when I re-applied the power to the UPS it started charging and after a short while the Pi started up.

It would be preferable if as soon as the power is re-established that if there is sufficient power in the battery the output is enabled.

Tom

organizedkaos, 2021-02-25:
I agree.  If we can configure it that might help with managing the power for the end user.

For me, if I can get battery voltage (at the minimum) from the uart data then I can actually script a capacity monitor process that could take into consideration voltage drop and allow me to choose to shutdown the pi or otherwise.

tomsimmons78, 2021-02-25:
@organizedkaos 

I'm not sure you can monitor the voltage, I believe it is using a buck converter so as the supply voltage drops it consumes more power to maintain the voltage.

When I came back to it at 12 hours it was showing the same voltage as the start.

Tom

organizedkaos, 2021-02-25:
I mean the source the voltage.  Yes, the boost convertor (or step-up) brings the voltage up to 5.xxx and maintains it but the source voltage (the 1s lipo) has a range of 4.2v (full) down to 3.4 (safe with a load).

When I tested the STA line the UPS didn't pulse until the capacity was at or below 5%.  Based on the voltage I was monitoring (at the source) this didn't happen until the lipo saw 3.2v with load (too low IMO).

If this UPS can't report source voltage then I would have to resort to using another hardware item to send the voltage range to the pi so that I can do the logic in python.

dominicdanna, 2021-03-10:
I'm having the same issue. Even when the UPS is charged to 100% (all 4 LEDs ON), reapplying power to the USB-C port does not force the pi to restart. The only way it seems to work is if you manually toggle the power switch on the UPS. This is not super useful. Anyone been able to get this listed feature to work properly?

new2thegame89, 2021-03-10:
Agree this is a big weakness. It also isn't clear from their description as they have it listed as the pi turning back on. Guess I'll need to look for another solution is the interim. Do you have any recommendations that you are familiar with? Thanks.

zify85, 2025-06-14:
I realize this is a very old thread but I have the newer V3 version with the 3.2 firmware and I have the same issue.  If I remove source power, shutdown the pi (via python), and reapply power quickly the pi never comes back on.  It's almost like the recharging logic to turn the pi back on isnt working.  It does power the pi if I turn off/on the UPS switch which tells me its some sort of logic error in the fw.  

Any solutions after 4 years of experience?