rcdrones/UPSPACK_V3 #11: RPi UPSPack Standard V3P - EXT BTN and exception when running UPS_GUI_demo.py

stevenfagg, 2021-04-09:
I recently bought the V3P version off of Amazon.
I finally found some information which linked me to this git repo with information regarding it.
However, there appears to be no information on the V3P version which comes with an EXT BTN header. I found this listing on xm680.com (https://www.xm680.com/goods/597160727755) for the exact version but I'm still none the wiser as to what I can do with the header and a button attached to the UPS.

Also, when running the UPS_GUI_demo.py demo application, it runs for about 6 seconds and then an exception is thrown:
```
pi@raspberrypi:~/UPSPACK_V3/UPS_GUI_py $ sudo python3 UPS_GUI_demo.py 
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/usr/lib/python3.7/tkinter/__init__.py", line 749, in callit
    func(*args)
  File "UPS_GUI_demo.py", line 14, in reflash_data
    version,vin,batcap,vout = test.decode_uart()
  File "/home/pi/UPSPACK_V3/UPS_GUI_py/upspackv2.py", line 22, in decode_uart
    self.uart_string = self.get_data(100)
  File "/home/pi/UPSPACK_V3/UPS_GUI_py/upspackv2.py", line 15, in get_data
    self.count = self.ser.inWaiting()
  File "/usr/lib/python3/dist-packages/serial/serialutil.py", line 590, in inWaiting
    return self.in_waiting
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 467, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
OSError: [Errno 5] Input/output error
```

When running minicom (and turning wrap on other wise all I see are $ signs at the far right of the window) I see the following continually being received:
`$ SmartUPS V3.2P,Vin GOOD,BATCAP 100,Vout 5250 $`
so in theory the UPS_GUI app *should* work but is failing.

Also, there doesn't seem to be anything in the above output which would identify if the button (plugged into EXT BTN) had been pressed.

a7hybnj2, 2021-06-05:
I too would like information on the external button. I put my V3P into an enclosure and would like to have a power button on the outside. I haven't yet probed anything and I haven't updated the firmware yet either.

If I figure anything out I will post it here.

rullinoiz, 2021-07-23:
i'm also trying to find documentation for the v3p model, google searches are coming out dry though 😕

rullinoiz, 2021-08-05:
i found the aliexpress listing for the v3p model and it comes with a self locking switch ( https://www.aliexpress.com/i/1005002374340811.html ). theres no information about the switch, but from how its plugged in, the ext btn header should act like an external switch, despite what is written on the board. my tests show that as long as SW+ and ground are shorted it will act as if the switch on the board is at the on position

a7hybnj2, 2021-08-05:
@rullinoiz Interesting looks like you need a latching button then. Also looks like black and red means you could have a button with LED. Neat.

The project I was working on that used the battery has been stagnant for a few weeks but when I pull it back out I will plug something in and see what happens.

a7hybnj2, 2021-09-05:
So I can confirm bridging the SW and GND pads does power the UPS. I thought I had read about a safe shutdown where when you turn the unit off that the pi completes its normal shutdown procedure and that doesn't appear to be the case. I will have to try to find where I originally read that information. Seems like there would need to be a few sensing lines between the UPS and PI for that to actually work.

rullinoiz, 2021-09-05:
@a7hybnj2 im pretty sure the safe shutdown feature is if the battery runs out, it'll send a signal through the serial pins to the pi telling it to shutdown, that way it doesn't corrupt anything

hoffmakl1961, 2021-12-11:
@ rullinoiz
for a safe shutdown GPIO18 (shutdown_pin = 18) on the Raspi and STA on the UPS board is used and not the serial interface (RX/TX -pins).

The serial interface is currently used for FW updates and gimmicks, which will not be used in later operation.

a7hybnj2, 2021-12-11:
@hoffmakl1961 Cool, I just looked at the python file and saw that. Very cool. I think it would be smart to implement that for my project.

It looks like it just prints maybe to stdout that it is shutting down. I think I will add an actual popup maybe with tkinter or something easy.

hoffmakl1961, 2021-12-12:
I use a UPS board RPI UPSPACK Standard V3P board with the firmware: SmartUPS V3.2P and shutdown_check.py. 

With this all basic functions of a UPS are available
1. the UPS has to make sure that the Raspi can shutdown safely. 
2. when the LiPo battery is recharged a bit, the output voltage is switched on again and the Raspi boots. (This was in the range of seconds for me!).  
3. the LiPo battery capacity must be at least so large, so that the Raspi can shut down safely. So the Raspi must not boot immediately when the power supply at the UPS is restored at the UPS and the LiPo battery capacity is too small to shutdown the Raspi safely.
4. additional functions are not required, which are realized e.g. with the serial port. (exception: firmware update.)

For me, the Raspi should always be in operation. What I have tested so far, everything seems to work.   

It would make sense to add it, but it is not part of the function of a UPS:
When the Raspi is powered up and SW+ is triggered: That a signal is sent to the Pi to shutdown_pin = 18 (GPIO18) and the Raspi shuts down.
When the Raspi is shutdown and SW+ is triggered: That the Raspi boots again. But only if the LiPo battery is charged enough to shutdown safely.

clarke-baxter, 2022-02-07:
I have a latching switch with LED bodged to the external switch connector. Anyone know the name of the plug type required for the external switch / LED socket?