rcdrones/UPSPACK_V3 #15: UPS GUI DEMO doesn't open

richsimpson, 2021-10-01:
Followed the instructions, have the earlier screen showing some data in terminal, including SmartUPS V3.2P, Vin GOOD,BATCAP 100, Vout 5250 $

[image]

But when double click the GUIDEMO it doesn't open anything ?

newpond, 2021-12-16:
@rich, did you fix this or do you need some help?

richsimpson, 2021-12-16:
> 

I didn't fix it, I gave up but happy to look again if you have any tips?

newpond, 2021-12-16:
hi rich,
can you enter
ls -l /dev

and confirm which serial ports are linked to which tty ports

eg serial0 -> ttyAMA0

have you added the +x permission to the UPS_GUI_demo.py file?

if not enter
chmod +x UPS_GUI_demo.py

then try running it again. It should bring up a popup asking you how you want to execute, in terminal etc. just hit the execute button.
let me know what you get
cheers
Jon

richsimpson, 2021-12-17:
Hi and thanks for the info. 

At step 4 of the README I have serial0 -> ttyAMA0
Reboot then Step 6 I get Some data back that seems to continue counting down the page with "$ SmartUPS V3.2P,Vin GOOD,BATCAP 100,Vout 5250 $" over and over. Its just off screen but maximised and I can see that happening so seems good.

At step 7 when I double click the UPS_GUI_demo.py, I get the query how to run, I select execute in terminal and a box flashes on, disappears and left with the normal desktop. If I select EXECUTE no box flashes and nothing opens.

If I go to terminal and copy that code you supplied I get "chmod: cannot access 'UPS_GUI_demo.py': No such file or directory
"

The file is in /home/pi/UPSPACK_V3/UPS_GUI_py and its called "UPS_GUI_demo.py"

richsimpson, 2021-12-17:
Just filmed the screen to see what the box was that opens and closes instantly. 
some permission denied.

[image]

newpond, 2021-12-17:
HI Rich, 
Can you confirm what raspberry pi model you are using?

How are you connecting to the desktop environment (ie via screen with mouse and keyboard, RDP or VNC) ?

What user are you logged in as when you run the gui demo program?

Can you also show the results of the following commands

from the UPS demo folder (/home/pi/UPSPACK_V3/UPS_GUI_py
ls -al

and from the /boot folder

more config.txt
or 
sudo more config.txt
(if the above command says you don't have permission)

cheers
Jon

richsimpson, 2021-12-18:
Hi Jon

Its a Pi4 8gb running debian10 buster
I am logged in as Pi
Connection is via VNC

Not sure I follow you completely. I opened the folder then opened termianl and ran the ls -al command. I get 

**start of pasting**
total 276
drwxr-xr-x 21 pi   users  4096 Dec 17 11:55 .
drwxr-xr-x  4 root root   4096 May 29  2021 ..
-rw-------  1 pi   users    56 Dec 17 11:55 .Xauthority
-rw-------  1 pi   users 11785 Dec 18 09:25 .bash_history
-rw-r--r--  1 pi   users   220 May  7  2021 .bash_logout
-rw-r--r--  1 pi   users  3523 May  7  2021 .bashrc
drwxr-xr-x  6 pi   users  4096 Oct  1 17:02 .cache
drwx------ 14 pi   users  4096 Oct  1 17:09 .config
drwx------  2 pi   users  4096 May 29  2021 .cups
drwx------  3 root root   4096 Nov  1 09:14 .dbus
-rw-r--r--  1 pi   users    35 Jul 13 12:45 .dmrc
drwx------  3 pi   users  4096 May  7  2021 .gnupg
drwxr-xr-x  3 pi   users  4096 May  7  2021 .local
drwx------  3 pi   users  4096 May 29  2021 .pki
-rw-r--r--  1 pi   users   807 May  7  2021 .profile
-rw-r--r--  1 pi   users    66 Jul 13 12:35 .selected_editor
drwx------  3 pi   users  4096 May 30  2021 .vnc
-rw-r--r--  1 pi   users   215 May 30  2021 .wget-hsts
-rw-------  1 pi   users  2349 Dec 17 11:55 .xsession-errors
-rw-------  1 pi   users  2349 Dec 17 11:46 .xsession-errors.old
drwxr-xr-x  2 pi   users  4096 May  7  2021 Bookshelf
drwxr-xr-x  2 pi   users  4096 May  7  2021 Desktop
drwxr-xr-x  2 pi   users  4096 May  7  2021 Documents
drwxr-xr-x  2 pi   users  4096 Oct  7 16:09 Downloads
drwxr-xr-x  2 pi   users  4096 May  7  2021 Music
drwxr-xr-x  2 pi   users  4096 May  7  2021 Pictures
drwxr-xr-x  2 pi   users  4096 May  7  2021 Public
drwxr-xr-x  2 pi   users  4096 May  7  2021 Templates
drwxr-xr-x  6 pi   users  4096 Nov  9  2020 UPSPACK_V3
drwxr-xr-x  2 pi   users  4096 May  7  2021 Videos
drwxr-xr-x  2 pi   users  4096 Jul 13 12:36 duckdns
-rw-r--r--  1 pi   users 14750 May 29  2021 get-docker.sh
-rwxr-xr-x  1 pi   users 18075 May 30  2021 install
-rw-r--r--  1 pi   users 18075 May 30  2021 install.1
-rw-r--r--  1 pi   users  7653 Sep 18 12:26 install.sh
-rw-r--r--  1 root root   8536 May 29  2021 installer.sh
-rw-r--r--  1 root root  67096 May 30  2021 openmediavault-omvextrasorg_latest_all5.deb
**end of pasting**

if i go to filesystem root, then open the boot folder and paste the sudo more config.txt i get a file/folder error:

more: stat of config.txt failed: No such file or directory

Is that what I was supposed to do, and if so then is that my issue, a missing item?

newpond, 2021-12-18:
Hi Rich
This is a pre build pi image I take it and not a manually installed Debian load?

It looks like you have run the ls -al command in the pi home directory and not in the /home/pi/UPSPACK_V3/UPS_GUI_py/ directory.
You mentioned you opened the folder and then opened terminal. I am guessing you meant you opened the folder in file manager, which is incorrect. this is all done with in terminal itself. (nothing to do with file manager).

please  open terminal (which if you are logged in as pi will open in the /home/pi directory automatically)

then type cd /home/pi/UPSPACK_V3/UPS_GUI_py

then type 
ls -al

what I am looking to see, is the permissions and ownership of the UPS files

the config.txt file is not in the root directory, it is in the /boot directory

so please type 
cd /boot
more config.txt
(or sudo more config.txt if that gives a permission error, which it shouldn't if you are using a pre build image)

rgds
Jon

richsimpson, 2021-12-20:
Thanks again JOn, NOt a great Linux user....

On the first command I get 👍 pi@raspberrypi:~/UPSPACK_V3/UPS_GUI_py $ ls -al
total 20
drwxr-xr-x 3 pi users 4096 Oct  1 17:17 .
drwxr-xr-x 6 pi users 4096 Nov  9  2020 ..
-rwxrwxrwx 1 pi users 3058 Nov  9  2020 UPS_GUI_demo.py
drwxr-xr-x 2 pi users 4096 Oct  1 17:17 __pycache__
-rwxr-xr-x 1 pi users 2740 Nov  9  2020 upspackv2.py

Second command (without sudo) does bring some feedback, here it is with the sudo command too:

pi@raspberrypi:/boot $ sudo more config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=16

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Enable DRM VC4 V3D drive
dtoverlay=vc4-kms-v3d
max_framebuffers=2
arm_64bit=1

# For 2020-08-20 - Raspberry Pi OS and newer versions
enable_uart=1
dtoverlay=disable-bt

Does that mean anything?
Cheers
Rich

newpond, 2021-12-20:
Hi Rich,
You're welcome.
All of that looks correct as it happens.

Can you open terminal and try the following

sudo raspi-config

this will open a configuration window

select
interface options
then
serial port

select "no" for the first option (to do with login shell)
and then "yes for the second option (to enable the serial hardware itself)

then reboot if it asks you to 

and then try running the UPS demo program again.

Rgds
Jon

richsimpson, 2021-12-20:
Thanks Jon, Ok I set as you said, I think it was already that way to be honest but rebooted and...

Problem solved, its working. 

Superb and thank you so much for your help on this. Top job!

newpond, 2021-12-20:
Hi Rich,
That's great news and glad I could help.
Just FYI, it will be that change you made in Raspi-config that fixed it.
The reason being that technically the guide you followed in the UPS documentation is wrong.
This is because when you added the lines
enable_uart=1
dtoverlay=disable-bt
to the config.txt file as stated in point 3 of the installation guide you "automatically" enabled the serial logging to shell function, which is what caused the problem. (known issue).
The documentation should really state only to add the disable-bt line and then to go into Raspi-config and made the changes I mentioned.
I am guessing that when you entered the serial port config, the first item (serial logging to shell) was highlight on "yes" and you then selected "no", which turned it off. The next page re turning on the serial hardware would also have been highlighted on "yes" which you left as it was! 
When you rebooted, it disabled the serial logging to shell option and hey presto it started working properly!

Glad its all working for you and that I could help.

Incidentally, it is possible on the raspberry pi 4 to use the UPS demo function without actually disabling and losing the bt (bluetooth) functionality. This is because the Pi4 actually has multiple uarts onboard, which things like the Pi zero doesn't have. I have both working on mine.
Let me know if this is something you require and I can help with this also. (not worth mucking around with however if you don't have the need for Bluetooth)

kind regards
Jon