rcdrones/UPSPACK_V3 #16: Does this actually need the uart connection

newpond, 2021-12-04:
I have been playing with this for quite some time now and find it a pain to have to lose the bt connection on my pi zeros (or fix the clock speed and move it to the uart port) or to have to add an additional uart port on my pi4s ( the zeros don't. support this).

However looking through the documentation and the python files, form what I can tell the uart communication is only used for the ups demo program (in order to pull the stats, battery levels etc) from the ups board.

However the actual automatic shutdown and start up feature appears to be built in the UPS board itself and is not configurable (or appear to be dependant on any uart communications) and purely uses the BCM pulse GPIO 18 to shutdown the device before disconnecting the power to the pi (automatic startup being done by enabling the power to the board again). 

Can someone confirm my assumptions are correct, that is if I don't need to use the ups demo program, the only connections "actually" required, are the gpio18 and the power (either via the usb-c port or via the power header on the ups board).
thanks
Jon

hoffmakl1961, 2021-12-11:
In normal operation: 
For shutdown_check.py only shutdown_pin = 18 is used. 

The serial interface is only required for a firmware update and python programs that are only used for demo purposes.

newpond, 2021-12-15:
Hi hoffmakl1961,
Thanks for your reply. I thought this might be the case.
Does the UPS pull GPIO18 low to shutdown the pi safely? If that's the case, can I also wire my shutdown switch to this pin? (it is a momentary push switch that shorts to ground, currently connected to gpio 26 (with shutdown overlay set in config.txt).
I can see in the UPS shutdown script it talks about pulses, but not really sure how this works! 

The shutdown script presumably only uses gpio18 for shut down purposes and once shutdown, cuts the power to the 5v connection to the Pi, and starts the pi back up by reconnecting the 5v again (once the main power has been restored and there is the right level of charge in the battery). I take it that's correct right?

Also one other thing that confuses me, is that the power wires between the UPS and the Pi are stated to require 24awg silicon cables due to the current required to power the pi (as I am using). How come the battery that was supplied with the UPS is clearly not using this gauge of cable (much thinner)? Surely if running on battery power, it will require the same current throughput as if on mains power?

I appreciate your help on this
cheers
Jon

hoffmakl1961, 2021-12-17:
Hi Jon,
I haven't found any documentation (schematic etc.) on borad yet and my python knowledge is limited as well. 

But what I see in the script:
GPIO18 is defined with a pull-down resistor. _GPIO.setup(shutdown_pin, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)_ 
I would not short GPIO18. The signal is pulled HIGH from the UPS board and the output current is limited by the internal pull-down resistor in the Raspberry. If you short this resistor with your push button, the output current will probably  
not be limited anymore and your UPS board (pin STA will be overloaded and defective afterwards.) Why don't you want to use GPIO26 as another possibility for a shutdown?     

In the while - loop pulsetime is set to 1 and the loop is exited when pulsetime is 2 or 3. 
Inside the while - loop there is a 2nd while - loop which checks if the shutdown_pin = HIGH. 
_GPIO.input(shutdown_pin) == GPIO.HIGH_  
If yes, plusetime is increased by 1  _plusetime += 1_ and with _break_ the loop is left and the shutdown is started. 
_os.system("sudo shutdown now -h")_
(Does the UPS pull GPIO18 low to shutdown the pi safely? No, responds to a high)

The script "only" shuts down the Raspberry, nothing more. 
The control itself is implemented on the UPS board. Depending on the battery capacity, the STA - pin (which is connected to GPIO18 on the Raspberry) sends the signal, as described above and after some time (when hopefully the Raspberry is safely stutdown) the power supply to the Raspberry is interrupted. When the battery capacity increases again, the board switches the power to the Raspberry back on and the Raspberry can boot. 

Wiring 
The cable between LiPo battery and UPS board with PH 2.0 connector is very short, so the total resistance on this connection should be small. But since the battery voltage is only about 3.7V and an output voltage of about 5V is required for the Raspberry. 
A step-up voltage regulator must be present on the UPS. This means that the current on the battery cable is much higher than the current on the cable to the Raspberry. 
My measurement showed: 
at an output voltage of 5.1V with a load of about 1.3A, I had a battery current of about 2.5A, at 3.7V.  
At a max. load of 2.5 ... 3A at 5.1V, a considerably higher battery current should be required. 

The assumption will be that the cable between UPS board and Raspberry could be considerably LONGER than the battery cable. 

I ordered the UPS - Board without battery and use 2x 18650 LiPo battery. As cable I use a 18 AWG cable, between battery - UPS - board and between UPS - board and Raspberry. I soldered the cables on the UPS board.
On the Raspberry I use the provided connector strip. 

Probably too much security, but I want a stable Raspberry.
cheers

newpond, 2021-12-17:
hi hoffmakl1961,
thanks for the above.

That makes sense re not using my switch on gpio 18, the only reason I thought about it was to reduce adding extra config and to free up a gpio pin that essentially is doing the same thing (ie shutdown).
Out of interest, should I use a resistor on my switch that connects to gpio 26? My understanding was that because I am pulling low rather than high, I shouldn't need one.

In regards to the pulse side of things, I'm still a bit confused. Does the UPS keep the gpio held low and then the script detects the it goes high and runs the shutdown command? And is the UPS sending actual pulses and if so in what form?
To add to my confusion the script actual uses the term "plusetime" rather than "pulsetime" (which I only just noticed) :-)

Re the cabling:
The battery that came with the UPS (10000mah) has much thinner gauge than 24awg which doesn't seem to make sense if as you say, it needs more current using the battery than it does when on mains power (unless as you say, they thought the length of the power cables would be quite long, but, saying that, if mounted under the RPI as appears the way it was designed to be installed, then not likely).
I bought an additional battery from amazon (3700mah, for testing) and that has REALLY thin cables, implying off the shelf stuff is not going to be up to the job, unless the cables are replaced with thicker ones...which I have done with the 10000mah one that was shipped with it.

Hope the above makes sense

cheers
Jon