rcdrones/UPSPACK_V3 #24: Battery type is unclear

ruidscampos, 2022-07-20:
Hoping anyone here is able to help me regarding battery type.
There is a reference to LiPo of different capacity and then to 3.7V 18650/21700/21650 battery pack, so long as the batteries are connected in parallel.

Just want to be sure I'm not messing it up by connecting a Li-Ion 3.7V 1S2P pack to it, was looking at something like this:
https://cpc.farnell.com/ansmann/2447-3033-20/battery-pack-li-ion-1s2p-3-7v/dp/BT06446?mckv=sXL3yKkm6_dc|pcrid|224680288544|kword||match||plid||slid||product|BT06446|pgrid|48556615353|ptaid|pla-436764189069|&CMP=KNC-GUK-CPC-SHOPPING&s_kwcid=AL!5616!3!224680288544!!!network}!436764189069!&gclid=Cj0KCQjwz96WBhC8ARIsAATR252R_lwm6jRPx4bX4pSn2kTy-qxemMnuPBVjeihq2VUZjKyOltzdTQwaAiBEEALw_wcB

Thanks in advance.

rcdrones, 2022-07-21:
connect to ups with a Li-Ion 3.7V 1S2P  is OK! no problem .

https://cpc.farnell.com/ansmann/2447-3033-20/battery-pack-li-ion-1s2p-3-7v/dp/BT06446?mckv=sXL3yKkm6_dc|pcrid|224680288544|kword||match||plid||slid||product|BT06446|pgrid|48556615353|ptaid|pla-436764189069|&CMP=KNC-GUK-CPC-SHOPPING&s_kwcid=AL!5616!3!224680288544!!!network}!436764189069!&gclid=Cj0KCQjwz96WBhC8ARIsAATR252R_lwm6jRPx4bX4pSn2kTy-qxemMnuPBVjeihq2VUZjKyOltzdTQwaAiBEEALw_wcB

ruidscampos, 2022-07-24:
Thanks @rcdrones !

closed 2022-07-24 by ruidscampos