rcdrones/UPSPACK_V3 #27: UART connection using UPS2() usually returns an OSError in python

bmosley, 2022-12-30:
nearly every time i run the simple script i hit errors:

```python
test = UPS2("/dev/ttyAMA0")

def log_data():
    version,vin,batcap,vout = test.decode_uart()
    batcap_int = int(batcap)

    print(f"UPSv3 ({version})")
    print(f"Battery Capacity: {batcap_int}%")
    print(f"Output Voltage: {vout} mV")

    if vin == "NG":
        print("USB input adapter: NOT connected!")
    else:
        print("USB input adapter: connected!")

```
i have a pi oled using I2C but there shouldnt be any issues reading the UART serial data.

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/serial/serialposix.py", line 549, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
OSError: [Errno 5] Input/output error

```

TinkerBridget, 2022-12-30:
Even the demo only worked intermittently for me, I kind of gave up on any of the communication features on that one

TinkerBridget, 2022-12-30:
The new Waveshare 3s one should work better, I'm just playing with one now, but still very n00bish with Linux and python